arduino/uno-r4-library-compatibility #2: PubSubClient (FAIL)

woolseyj, 2023-06-29:
Compilation of the [PubSubClient](https://github.com/knolleary/pubsubclient) library fails with the following error:
"error: 'strnlen' was not declared in this scope" and suggests using 'strlen' as an alternative.

It appears the strnlen() function is not defined for the Renesas architecture.

I tried to compile the [mqtt_basic](https://github.com/knolleary/pubsubclient/tree/master/examples/mqtt_basic) example sketch with a few modifications related to changing the _Ethernet_ library for the _WiFiS3_ library.

The verbose log file is enclosed.

[mqtt_basic_compile_log.txt](https://github.com/arduino/uno-r4-library-compatibility/files/11906291/mqtt_basic_compile_log.txt)

tony2feathers, 2023-07-30:
I made a fork of PubSubClient which addresses these issues. I have tested it and have managed to get it to connect to my wifi and connect to the test.mosquitto.org mqtt broker. I am having problems with my local broker for some reason, and I'm unsure what the cause is, but the new forked library works as far as I can tell. https://github.com/tony2feathers/pubsubclient_UnoR4

woolseyj, 2023-09-08:
The [issue](https://github.com/arduino/ArduinoCore-renesas/issues/18) was resolved in the ArduinoCore-renesas library.  The PubSubClient library is now working properly with the Arduino R4 WiFi board.

closed 2023-09-08 by woolseyj

Hannes7eicher, 2023-09-11:
I added it [here](https://github.com/arduino/uno-r4-library-compatibility/pull/22).