arduino/uno-r4-library-compatibility #3: SoftwareSerial

schultzda, 2023-07-01:
Link to test program  -> https://docs.arduino.cc/tutorials/communication/TwoPortReceive
UNO R4 Wifi was configured to use the pins as defined in the program.  I was using two ultrasonic sensors (A02YYUW) and their RX and TX pins.
Fails to compile with reported error of  " Class SoftwareSerial has no member named 'listen'

I reinstalled the AVR core libraries to make sure that I had updated libraries.  I also drilled down to SoftwareSerial.h file on my computer to make sure that listen was included and it was.  

FAIL

Hope I did this correctly as this is my first report here.

phfbertoleti, 2023-07-14:
Regarding SoftwareSerial in Uno R4 Minima, I couldn't make it work using GPIOs 10 and 11. It seems nothing is being transmitted nor received using SoftwareSerial in Uno R4 boards in these GPIOs

Hsubtnarg, 2023-07-15:
I have the same issue using pin 10,11. did you get it working with any other pins?

schultzda, 2023-07-16:
I just received the response from Arduino yesterday.  I have not had a
chance to try different pins.  Since I need two Serial ports for the
project that I’m trying to create and it is beginning to appear that the
UART is reaching its end of life, I’m thinking of switching over to using
PWM instead.  The sensor I’m using offers either serial or PWM.  Needless
to say the project has moved to the back of the list for now.

Hsubtnarg, 2023-07-16:
I tryed pins 8 and 9 and it seems to work although now Im getting corruption on the messages Im sending, but I get the same with the hardware serial, so there is still something wrong

schultzda, 2023-07-16:
One of the reasons why I have put my project on the back burner is because
the New R4 chips need some time to work out the unforeseen “bugs.”  I’m
sure that Libraries are not considered to be low hanging fruit on their
development tree and thus not the first problems to be resolved.  I have
set a time line of several months before I revive my project to give them
the time they need to address all the issues.  While they claim “Drop in
Compatibility” with the R3, since many Library files are developed by
3rd parties it could be some time before many of the “bugs” are worked out.
Just a thought to consider in my development process.

alranel, 2023-07-17:
Hello!
Not all pins can be used for SoftwareSerial. See this note in the examples for SoftwareSerial on UNO R4:

```
// Note any pin can be used for TX, but only the following pins
// can be used for RX:
// D0, D1, D2, D3, D8, D14, D15, A1, A2, A3, A4, A5
```

closed 2023-07-17 by alranel

Hsubtnarg, 2023-08-02:
I have been doing some more investigating on this and looks like the Uno R4 is not sending the stop bit correctly. For 9600baud it should be about 120us, but it is only 62us.
[image]

This is cause the frame to drift and causing errors on the reciver. This is the same code running on a Lenardo.
[image]

Rojasoterom, 2023-11-28:
For the record:

I use SoftwareSerial on R4 Wifi, using ports 0,1 (yes, ports of Serial1). When i disconnect USB C, Arduino capture the information without problems (im sending information via MQTT to see the data).

ShortArrow, 2024-01-22:
> Hello! Not all pins can be used for SoftwareSerial. See this note in the examples for SoftwareSerial on UNO R4:
> 
> ```
> // Note any pin can be used for TX, but only the following pins
> // can be used for RX:
> // D0, D1, D2, D3, D8, D14, D15, A1, A2, A3, A4, A5
> ```

Add link to [Example of SoftwareSerial on ArduinoCore-renesas](https://github.com/arduino/ArduinoCore-renesas/blob/main/libraries/SoftwareSerial/examples/SerialEcho/SerialEcho.ino)