arduino/uno-r4-library-compatibility #11: Arduino_NineAxesMotion (COMPILATION PASS, HARDWARE FAIL)

framoc0991, 2023-07-29:
I am testing the Arduino_NineAxesMotion library 

https://github.com/arduino-libraries/Arduino_NineAxesMotion/tree/master

suggested for the 9 Axis Motion shield (in arduino Webpage).
Compilation seems ok, no errors.
When running the examples provided in the library (started with the Accelerometer example), the output data on the serial monitor lags consistently. No effects when changing the serial communication baudrate, thus I suppose it has problem in updating sensor data in these lines before the serial prints

if (updateSensorData)  //Keep the updating of data as a separate task
  {
    mySensor.updateAccel();        //Update the Accelerometer data
    mySensor.updateLinearAccel();  //Update the Linear Acceleration data
    mySensor.updateGravAccel();    //Update the Gravity Acceleration data
    mySensor.updateCalibStatus();  //Update the Calibration Status
    updateSensorData = false;
  }

Looking on the Arduino forum, I have a bad feeling about the I2C communication, but I do not have hardware to further test it.

sidscopexx, 2023-12-21:
Have you found a solution to the problem? I have the same problem and can't get any further.

framoc0991, 2023-12-22:
> Have you found a solution to the problem? I have the same problem and can't get any further.

Dear sidscopexx,
I did not have worked to the project since July, thus my answer could be outdated. Anyway, at the following link you can find the discussion we had on the Arduino forum in July.

https://forum.arduino.cc/t/i2c-setclock-does-not-appear-to-be-standard-way-to-setclock/1147122/17

In summary, it seems that in the "old" wire library there was a compatibility issue that did not allow to change the clock speed to adapt the I2C communication to the new renesas platform (sorry if I am not speaking correctly, but I am not an firmware programmer). I sincerely do not now if the current Wire library already solved that problem. Try to 
Wire.setClock(I2C_MASTER_RATE_FAST);
after
Wire.begin();
If the compatibility issue is included in the new Wire lib, it should work.
Hope it helps

Hannes7eicher, 2024-08-22:
Apologies for the late reply! I just ran some tests using the [9 Axis Motion Shield](https://docs.arduino.cc/hardware/9-axis-motion-shield/) and the UNO R4 WiFi and so far everything works fine. I am not sure if [Wire](https://github.com/arduino/ArduinoCore-renesas/tree/main/libraries/Wire) was updated to fix any previous issues.

Let me know if there are any issues remaining, in that case I'll set aside some time to investigate.