arduino/uno-r4-library-compatibility #13: Joystick (FAILED)

cbrunnem, 2023-08-05:
joystick library from here fails to complile. https://github.com/MHeironimus/ArduinoJoystickLibrary/tree/master/examples/JoystickButton

verbose log:
`FQBN: arduino:renesas_uno:unor4wifi
Using board 'unor4wifi' from platform in folder: C:\Users\cbrun\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2
Using core 'arduino' from platform in folder: C:\Users\cbrun\AppData\Local\Arduino15\packages\arduino\hardware\renesas_uno\1.0.2

loading library from c:\Users\cbrun\Documents\Arduino\libraries\test: invalid library: no header files found
Detecting libraries used...
"C:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DNO_USB -DBACKTRACE_SUPPORT -DARDUINO_UNOR4_WIFI -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"C:\\Users\\cbrun\\AppData\\Local\\Temp\\arduino\\sketches\\C57DBDE7ECCA062C22C2A91095048BCB/JoystickButton.ino\"" -DARDUINO_UNOWIFIR4 -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb "@C:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\UNOWIFIR4/defines.txt" -DCFG_TUSB_MCU=OPT_MCU_RAXXX "-IC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2/cores/arduino/tinyusb" "-IC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino" "-IC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\UNOWIFIR4" "-iprefixC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2" "@C:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\UNOWIFIR4/includes.txt" "C:\\Users\\cbrun\\AppData\\Local\\Temp\\arduino\\sketches\\C57DBDE7ECCA062C22C2A91095048BCB\\sketch\\JoystickButton.ino.cpp" -o nul
Alternatives for Joystick.h: [Joystick@2.1.1]
ResolveLibrary(Joystick.h)
  -> candidates: [Joystick@2.1.1]
"C:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -Os -g3 -fno-use-cxa-atexit -fno-rtti -fno-exceptions -nostdlib -DF_CPU=48000000 -DNO_USB -DBACKTRACE_SUPPORT -DARDUINO_UNOR4_WIFI -std=gnu++17 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsigned-char -ffunction-sections -fdata-sections -fmessage-length=0 -fno-builtin -w -x c++ -E -CC -DARDUINO=10607 "-DPROJECT_NAME=\"C:\\Users\\cbrun\\AppData\\Local\\Temp\\arduino\\sketches\\C57DBDE7ECCA062C22C2A91095048BCB/JoystickButton.ino\"" -DARDUINO_UNOWIFIR4 -DARDUINO_ARCH_RENESAS_UNO -DARDUINO_ARCH_RENESAS -DARDUINO_FSP -D_XOPEN_SOURCE -mthumb "@C:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\UNOWIFIR4/defines.txt" -DCFG_TUSB_MCU=OPT_MCU_RAXXX "-IC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2/cores/arduino/tinyusb" "-IC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated" "-IC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\cores\\arduino" "-IC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\UNOWIFIR4" "-Ic:\\Users\\cbrun\\Documents\\Arduino\\libraries\\ArduinoJoystickLibrary-master\\src" "-iprefixC:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2" "@C:\\Users\\cbrun\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\renesas_uno\\1.0.2\\variants\\UNOWIFIR4/includes.txt" "C:\\Users\\cbrun\\AppData\\Local\\Temp\\arduino\\sketches\\C57DBDE7ECCA062C22C2A91095048BCB\\sketch\\JoystickButton.ino.cpp" -o nul
Alternatives for PluggableUSB.h: []
ResolveLibrary(PluggableUSB.h)
  -> candidates: []
In file included from c:\Users\cbrun\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/Joystick.h:24:0,
                 from C:\Users\cbrun\AppData\Local\Temp\.arduinoIDE-unsaved202375-19284-st8wk2.qgsdj\JoystickButton\JoystickButton.ino:14:
c:\Users\cbrun\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/DynamicHID/DynamicHID.h:37:12: fatal error: PluggableUSB.h: No such file or directory
   #include "PluggableUSB.h"
            ^~~~~~~~~~~~~~~~
compilation terminated.

Using library Joystick at version 2.1.1 in folder: C:\Users\cbrun\Documents\Arduino\libraries\ArduinoJoystickLibrary-master 
exit status 1

Compilation error: exit status 1
`

per1234, 2023-08-27:
There is also a request for adding support for the UNO R4 boards to the library here: https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/280

Hannes7eicher, 2023-09-07:
The Joystick library has been added [here](https://github.com/arduino/uno-r4-library-compatibility/pull/20).

closed 2023-09-07 by Hannes7eicher