arduino/uno-r4-library-compatibility #17: WiFi101/ATWINC1500 shield is not compatible

JAndrassy, 2023-08-30:
Arduino WiFi101 shield and Adafruit ATWINC1500 shield is not compatible out of box, because it requires interrupt capable pin 7.
It works when pin 7 is shorted to pin 2 (or 3) and in sketch pins are set as
```
  WiFi.setPins(10, 2, 5, -1); // -1 is en pin not used
```
Adafruit ATWINC1500 breakout boards must wire IRQ pin to Uno R4 interrupt pin  2 or 3 instead of the recommended pin 7

Hannes7eicher, 2024-08-22:
Apologies for the late reply and thanks for taking the time to investigate. Just out of curiousity is there a reason using a WiFi shield with the UNO R4 (presumably the Minima) instead of using the UNO R4 WiFi?

JAndrassy, 2024-08-22:
I don't know. It has more stable and faster WiFi, SD card slot, prototype area. Different projects have different requirements.