arduino/uno-r4-library-compatibility #18: my libraries ArduinoOTA, WiFiEspAT, EthernetENC, TelnetStream, StreamLib (PASS)

JAndrassy, 2023-08-30:
all are in category Communication

WiFiEspAT and EthernetENC worked without changes

ArduinoOTA 1.0.11 has new support for Uno R4 and for the WiFiS3 library
TelnetStream 1.2.5 has new support for the WiFiS3 library
StreamLib 1.3.2 has a patch for the Renesas core

Hannes7eicher, 2023-09-07:
Hi @JAndrassy!

Awesome! I've added them all in this PR [here](https://github.com/arduino/uno-r4-library-compatibility/pull/20).

Thank you for contributing!

closed 2023-09-07 by Hannes7eicher

FMM1961, 2024-01-08:
I can not get Telnetstream to work with Arduino UNO R4 WiFi which uses the WiFiS3 library. The comment above on Aug 30 states it now supports that library. I don't get a fault when compiling, and I can establish a telnet connection but the .read and .print statements do nothing.