arduino/uno-r4-library-compatibility #28: Serial1.write(buf, len) (FAIL)

SeanMWX, 2024-02-18:
Problem description:
I am encountering an issue with **Serial1.write(buf, len)** where the number of bytes received does not match the expected length during Serial1 communication between two Arduino devices with TX and RX pins cross-connected.

Setup for Arduino 1:
1. Initialized a buffer **"uint8_t au8Buffer[64]"** and assigned values to the first eight elements.
2. Attempted to send the first eight elements via **"Serial1.write(au8Buffer, 8)"**, expecting to transmit 8 bytes.

Setup for Arduino 2:
1. Configured to listen for incoming Serial1 communication.
 
Expected behavior:
Arduino 2 should receive exactly 8 elements when Arduino 1 sends a length of 8.

Observed behavior (See figures below):
1. When sending with **"len = 8"**, Arduino 2 receives only 7 elements.
[image]

2. Increasing the send length to **"len = 9"** results in the correct reception of 8 elements. 
[image]

**This behavior is not consistent with my understanding of the Serial.write() documentation and intended functionality.**

Additional:
Arduino IDE version: 2.3.1
Code for Arduino 1:
``` C++
uint8_t au8Buffer[64];

void setup() {
  Serial.begin(9600);
  Serial1.begin(9600);
  au8Buffer[0] = 1;
  au8Buffer[1] = 3;
  au8Buffer[2] = 0;
  au8Buffer[3] = 1;
  au8Buffer[4] = 0;
  au8Buffer[5] = 4;
  au8Buffer[6] = 21;
  au8Buffer[7] = 22;
}

void loop() {
  Serial1.write(au8Buffer, 8);  // len = 8 only sent first 7 elements ...???
}
```
Code for Arduino 2:
``` C++
void setup() {
  Serial.begin(9600);
  Serial1.begin(9600);
}

void loop() {
  if(Serial1.available()) {
    Serial.println(Serial1.read());
  }
}
```

Hannes7eicher, 2024-08-22:
Apologies for the late reply! I ran your examples and I get the same result.  I added a `delay(1000)` to make sure the buffer is not overflowing and that fixed the issue.

SeanMWX, 2024-08-24:
> Apologies for the late reply! I ran your examples and I get the same result. I added a `delay(1000)` to make sure the buffer is not overflowing and that fixed the issue.

Thank you for the reply, and if that fixed the issue, then I think we can close the issue. :)